arduino/arduino-pro-ide #440: IntelliSense doesn't work without #include <Arduino.h>

chovanj, 2021-02-19:
**Describe the bug**
IntelliSense doesn't work correctly. 
After adding #include <Arduino.h> it start working fine.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Arduino PRO IDE : 0.1.4.

kittaakos, 2021-02-19:
@chovanj, which board did you select? At the moment, the language features do not start until you pick a board.

chovanj, 2021-02-19:
> @chovanj, which board did you select? At the moment, the language features do not start until you pick a board.

[image]

kittaakos, 2021-02-19:
Do you see any errors in the console if you start the app from `CMD.exe`?

kittaakos, 2021-02-19:
Also, do you see an error like `Couldn't start client Arduino Language Server` (or similar) when you start the app?
You can see the notification center by clicking on this icon:
[image]

chovanj, 2021-02-19:
No, there are no messages in the status bar.

kittaakos, 2021-02-19:
Can you please enable the language server logs and share them with us?

----

Please follow the steps if you want to help us figure out what's wrong:
 - create a new sketch,
 - save it (so that you will find it later in your _Sketchbook_),
 -  open the preferences with <kbd>Ctrl</kbd>+<kbd>,</kbd>,
 - select _Enable language server logging_,
[image]
 - and pick a board; you can reselect the current one. (The last steps should trigger the language server restart.)

You will find a few log files in the sketch folder; could you please attach them to this issue? Thank you!

kittaakos, 2021-02-20:
By the way, did you install the platform for the Uno board?

Bad-ptr, 2021-02-21:
Confirm.
As for me, if Arduino.h is included in sketch(the simplest one), then in the error list the IDE tells me that it can not find stdlib.h which is included by Arduino.h.
Compilation of the sketch is successful.
Only autocompletion and error highlighting is affected.

chovanj, 2021-02-23:
> By the way, did you install the platform for the Uno board?

Hi, from this video it could be more obvious.

[image]

kittaakos, 2021-02-24:
> from this video it could be more obvious.

Could you please share the log files with us to make this more obvious? Thanks!