arduino/arduino-pro-ide #441: Missing Serial Plotter

sakorada, 2021-02-24:
**I would like to see Serial Plotter on Arduino Pro IDE very soon.
Thank you**

ubidefeo, 2021-02-24:
thank you for your interest, @sakorada 
Serial Plotter is in our roadmap after the Beta release

kittaakos, 2021-02-24:
Ref to old issue: https://github.com/arduino/arduino-pro-ide/issues/115

kittaakos, 2021-02-24:
Ref to old issue: https://github.com/arduino/arduino-pro-ide/issues/171

kittaakos, 2021-02-24:
Additional requirements form [here](https://github.com/arduino/arduino-pro-ide/issues/171#issuecomment-725387387):

> The serial plotter is a very useful feature but it needs improving:
> Add an 'auto scale' checkbox so automatic scaling can be turned off.
> Add a zoom slider for manually adjusting the display scale.