arduino/arduino-pro-ide #443: Cannot see the full Arduino Pro IDE window on Linux (Ubuntu 20.04) laptop

LumitoLuma, 2021-02-25:
**Describe the bug**
I cannot see the full Arduino Pro IDE window on my Ubuntu 20.04 (GNOME) laptop. I also cannot maximize it.

**To Reproduce**
Steps to reproduce the behavior:
1. Download Arduino Pro IDE 0.1.4 on Ubuntu 20.04
2. Extract
3. Open `arduino-pro-ide`
4. See error

**Expected behavior**
Be able to fully resize the IDE window and maximize it.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Linux Mint (with GNOME desktop, based on Ubuntu 20.04.1 LTS)
 - Version: 20.1 Ulyssa

**Additional context**
I also cannot vertically resize the window

ubidefeo, 2021-02-25:
hi @LumitoLuma 
can you add your display resolution?

thanks

kittaakos, 2021-02-25:
Thank you for reporting it, @LumitoLuma 👍 This is a known issue and tracked under https://github.com/arduino/arduino-pro-ide/issues/337. I am closing this one as a duplicate.

closed 2021-02-25 by kittaakos

kittaakos, 2021-02-25:
On a Windows machine, the resolution is 2560x1440, and the app is clipped. One has to maximize it to see the status bar. We cannot rely on fix height/width.