arduino/arduino-pro-ide #444: Arduino Pro IDE GUI is gone on Windows 10

jetzoc, 2021-02-25:
**Describe the bug**
Opened up the Arduino Pro IDE (v0.1.4) and cannot see the GUI screen on Windows 10 machine. Splash screen shows up but there is no GUI interface. I downloaded the zip file opened the IDE once or twice but then switched to Linux and started to use the IDE there. Came back to Windows but now I don't see the GUI. I can hover over the taskbar and see a preview of the IDE but I do not see it at all. 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'arduino-pro-ide_0.1.4_Windows_64_bit' folder
2. Click on 'Arduino Pro IDE.exe'
3. View the splashscreen
4. No GUI seen but can see the taskbar view

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windows
 - Version: 10

**Additional context**
I'm sure it has something to do with my own machine, of course, given that I had the program under a folder that OneDrive wanted to back up. And, I didn't want it there anymore because OneDrive is annoying. So, I deleted that version because of the problem. I re-downloaded the zip, extracted it in my _Downloads_ folder, opened it from there but the same thing - no IDE in sight.

kittaakos, 2021-02-26:
Thanks for reporting it, @jetzoc. There should be a `config.json` file under `C:\Users\YOUR_NAME\AppData\Roaming\arduino-pro-ide`? Can you please share its content with us?

kittaakos, 2021-02-26:
By the way, do you use multiple screens for your Windows env? Thanks!

jetzoc, 2021-02-26:
Yes, of course, here are the contents. And, no, I don't use multiple screens; however, I do remember, changing the magnification levels on my display at one point but, I reverted the setting back to default. 

```
{
	"windowstate": {
		"isMaximized": false,
		"width": 968,
		"height": 781,
		"x": -20,
		"y": 1394
	}
}
```

kittaakos, 2021-02-26:
Thank you! Could you please delete that file and see if it fixes the issue? (This file stores the location of the app on the screen.)

jetzoc, 2021-02-26:
Okay, so I changed the ``is maximized`` parameter to ``true``. 
And, I was able to restore it. 

But, yes, I can delete the file as well. That works just the same

Thank you!

closed 2021-02-26 by jetzoc

kittaakos, 2021-02-26:
I am glad you could solve the issue.

I opened a follow-up here: https://github.com/arduino/arduino-pro-ide/issues/445